EdgeTX/edgetx #6568: TX15 Logging Causes Freez up

TB2185, 2025-09-05:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When Trying to Log on the TX15 Without the SD Card. It will Cause the radio to Freeze up when powering down. Only way to get out of it is to Pull the Battery. Looked for a Internal Loging option there wasnt any. so added an SD card. and that has fixed the issue.

### Expected Behavior

There Should be an Internal logging Option. 

### Steps To Reproduce

Turn on Data Loging with no SD Card in there . Start Loging. then turn off Radio and it will Freeze up. 

### Version

Other (Please specify below)

### Transmitter

Other (Please specify below)

3djc, 2025-09-06:
The system uses either internal sd or external. If you don't have an external SD, then logging is done on internal, like it is on other radio with internal sd. There is absolutely no need to the option you mention

Logging on internal SD works fine here, please share your sd content so that we can try to replicate

J-Sorenson, 2025-09-08:
We need more data, please:
Does it freeze on power-down every time you use logging, or is it intermittent?
Does it freeze if you power on then power down *without* enabling logging?
Are your log files present, and do they seem complete?
Have you tried clearing all of your logs and then trying again?
Most importantly: what are you logging?