EdgeTX/edgetx #6576: Wayland support

ask6483, 2025-09-10:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Running on wayland session results in:
```
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.
```
I have to `export QT_QPA_PLATFORM=xcb` explicitly to make it work again.

### Expected Behavior

Wayland session will be the default one in Gnome soon.

### Steps To Reproduce

1. Open Gnome on Wayland session
2. Run edgetx-companion

### Version

2.11.3

### Transmitter

RadioMaster Boxer

### Operating System (OS)

Linux

### OS Version

Debian Sid

pagrey, 2025-09-13:
Nothing about QT in the environment variables, XDG_SESSION_TYPE=wayland appears but I didn't add or edit anything extra myself.

J-Sorenson, 2025-09-16:
I grabbed Version 2.11.3 of the appimage, and I confirmed that it does not support Wayland at all.  It works fine on Fedora Linux 42 (Wayland default, but X11 fallback).  XDG_SESSION_TYPE=wayland, and the only QT env var is QT_IM_MODULE=ibus.  I get the warning, but the application starts up:
```
jim@framework:~/Apps$ ./EdgeTX_Companion_2.11.3-x86_64.AppImage
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[I] AppData::init():543 - Settings init with EdgeTX Companion 2.11 Saved version: 20B0302 Current version: 20B0302
[D] Translations::installTranslators():106 - Locale name: "en_US" language: "American English" country: "United States"
```

However, if I do set "export QT_QPA_PLATFORM=wayland", then I can reproduce your error:
```
jim@framework:~/Apps$ export QT_QPA_PLATFORM=wayland
jim@framework:~/Apps$ ./EdgeTX_Companion_2.11.3-x86_64.AppImage
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.

Aborted (core dumped)
```

Fedora Linux 43 removes X11 fallback, so the next patch release will certainly need the Wayland libraries included.

J-Sorenson, 2025-09-16:
This recent Qt thread shows how one person combined both xcb and wayland compositors in their appimage.  Looks to be a bit of a manual process (not default): https://forum.qt.io/topic/163015/problem-running-application-on-linux.-ubuntu-22.04-lts-wayland-platform.

pagrey, 2025-09-16:
I'm sure the issue will get resolved, until it does there are several workarounds. Obviously, you can run Xwayland, just because it isn't installed by default doesn't mean you can't use it. You can also follow the general [Build Instructions under Ubuntu 20.04](https://github.com/EdgeTX/edgetx/wiki/Build-Instructions-under-Ubuntu-20.04), this is what I've done and the result runs fine on Wayland. Allot of stuff doesn't work on Wayland so Xwayland seems like the way to go.

pfeerick, 2026-10-21:
Ok, so this issue was originally raised against 2.11, which uses QT5 (not QT6 as Raphael stated)... but that still supports wayland via QtWayland. I can launch 2.11.5, 2.12.4 and main branch Linux Companion without issue on Fedora 44 (GNOME/Wayland), Debian 13 (GNOME/Wayland) and KDE neon (Plasma 6.8). All three are basically clean systems with no extra compatibility wrappers or any tweaks done. I'm not sure about Fedora, but the Debian and KDE neon system have some x11 packages installed standard, so it seems even on "official" KDE OS it is expected that you will have some form of x11 present. Companion only breaks with the above errors on these systems if I **explicitly** set `export XDG_SESSION_TYPE=wayland`. 

Having said that, it seems like it is simply a missing packaging command to tell `linuxdeploy` to also bundle the wayland plugin, which removes the 

```
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
```

error, and does not seem to change compatability with a x11-only session, so #7930 should resolve this.

closed 2026-10-22 by pfeerick