EdgeTX/edgetx #6587: TBS Agent Lite 0.98 will not run

ror1948, 2025-09-14:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

when you try and run lua TBSAgentLite.lua nothing happens

### Expected Behavior

expect a menu to appear on screen

### Steps To Reproduce

press SYS
select Lua TBSAgentLite 098

### Version

2.11.3

### Transmitter

RadioMaster TX16S / TX16SMK2

closed 2025-09-14 by JimB40

ror1948, 2025-09-14:
0.98 is what is displayed on their website but thanks

[image]

pfeerick, 2025-09-14:
Looks like the links have finally been updated... 0.99 is there now as well as the version for ETHOS.

ror1948, 2025-09-14:
No i still see 98 at teamblacksheet website