EdgeTX/edgetx #6605: feat: Allow special functions to change color of customized switches

Forceu, 2025-09-25:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

The are situations where I would like to have an indicator light on my remote, e.g. for GPS lock or RX signal strength.

Currently as a workaround I assign a customizable switch an on and off colour (red/green) and use a special function to simulate a press, in order to change colour.

This has the disadvantage, that only two colours can be used. These two example are currently not possible:

- Having a red/yellow/green indicator for signal strength
- Alternating between red/off (blinking LED) and then turning green when searching for satellites and finally having a GPS lock.

### Describe the solution you'd like

If a special function was added to enter a colour code for a switch (or use a predefined colour), this would make indicator lights possible.
If there are concerns, that it would make it harder to distinguish the state of a button when manually setting the colour, a fourth type could be added to `None`/`Toggle`/`2Pos` named `Indicator` which only allows changing colour, but a button press would not do anything.

If there are concerns to inputting a colour as a parameter through a special functions, another option could be set more states for a switch with the `Indicator` state (e.g. 6 states) in the button config and simply pass the state number with the special function.

3djc, 2025-09-25:
We have allowed LUA to be used to pilot those in latest versions. LUA allows you much more control, you could gradually move from red to yellow then green for rx signal for example, or define a logic of gps lock, things that would be really hard to do using SF

Forceu, 2025-09-25:
Thanks for the quick reply! I didn't realise this is possible, if you want to, you can close this issue then.
Do you know the function to manipulate the custom switches with LUA by any chance? In the API reference for models I could not find any function in that regard: https://luadoc.edgetx.org/lua-api-reference/model

Forceu, 2025-09-25:
Thank you for the clarification!

closed 2025-09-25 by Forceu